githubdoe/DFTFringe #173: Save screen can place the 3D image in the wrong spot and it's remembered file path is not very useful

githubdoe, 2025-02-07:
If you use the file menu save image and do not have DFT Fringe screen maximized the 3D opengl image be placed in the wrong spot.
Also it uses the last saved mirror config path as the path where to save the image.  I find it more useful to have it rememer it's own path.  So recommend changing the saving of the last place it was saved as the remembered path.

I have fixed both issues in my personal build.   It turned out to be an easy fix.  However it has been suggested that the save image feature be eliminated because there are other ways to external to DFTFringe to get a saved image.  So I'm waiting to see what the consensus is.  If we want to keep the feature then I can make another branch with my fix and proceed from there.

gr5, 2025-02-08:
@atsju what do you think?  It does the same thing as any screen capture software including windows own built in "snipping tool".  I see no advantage to having it in the app.  People need to learn to use the snipping tool.  It's one of the most basic things that people should learn after they learn the difference between left and right click but before they learn about the "office" apps.

githubdoe, 2025-02-08:
Ha, Ha,  I don't know what the snipping tool is.  I have a vague memory of hearing about it once.    I don't much use the "office" (LIbreoffice) apps either.  Use mostly Chrome, DFTFringe, IrfanView, Fusion360, Davinci Resolve, Qt Creator, VS CODE.  And cuss at github.

gr5, 2025-02-08:
Just hit window key and type "screen" as in screen capture and it pops up.

gr5, 2025-02-08:
Actually I think you can hit the "print screen" button also.  There's also shift+print_screen and ctrl+print_screen and alt+printscreen.  Some of those capture the whole screen, some the active window, some let you lasso an area.

I installed some kind of screen capture software that was free (greenshot) and it redefined all those keys.  But by default I think they all run snipping tool.  I usually use snipping tool on windows.  On my linux machine which I use the most I also use "print screen" key a lot and so when I switch to windows I sometimes forget and use the same key without thinking.

atsju, 2025-02-08:
I use snipping tool daily. I would lean toward reducing DFTFringe feature set to ease maintenance and focus on things that mass realy needs. 

In my opinion removing this feature is better solution.

githubdoe, 2025-04-21:
I'm leaving the feature in.  Ready for testing or removal by others.

gr5, 2025-04-25:
works fine in build dalework