githubdoe/DFTFringe #174: Read outline fails to update the display with the new outline.

githubdoe, 2025-02-08:
Load an igram that has an outline.  Then press the read ouline button and load a different outline.  It looks like nothing happened but if you do something like press the left arrow or scroll the middle scroll while to update the display you will now see the new outline.  For some reason the display is not updated when the "Read outline" button is used to load an outline.

githubdoe, 2025-02-08:
I think the display will get updated if a  DrawBoundary() call is made at the end of the IgramArea::readoutlines function.

githubdoe, 2025-04-21:
I think this bug was corrected and is ready for testing.

gr5, 2025-04-25:
This works fine now in branch dalework.

atsju, 2025-06-26:
Should this be closed after latest releases ?

closed 2025-06-26 by gr5