githubdoe/DFTFringe #179: Add hot keys to main window.

githubdoe, 2025-03-18:
Recently I have found it useful to use a hotkey "I" on the main window to import interferograms without going to the file menu.  I have added extra functionality as well that depending on the preferences opens the most recent image file in a directory specified by the preferences.  In my case it is the download directory on my laptop.  That is because I'm importing images from my skysolve raspberry pi camera that I use to take igrams with.  Those images get placed in the download directly from the skysolve html web page.  

I find with that then I can quickly evaluate the current igram being shown in the camera to discover if the settings are good.  A future enhancement could be done to allow it to capture images from a stream as well.  

Anyway the code will be in my dalework branch soon.  Then we can discuss if and when it gets added to the main branch.

It might also be nice to use the hot key "w" to open the current wave front directory.  "o" to jump to the outline igram tab (maybe.)

gr5, 2025-03-18:
The standard key to open a file is CTRL+O and an igram would be the default file to open in my opinion but some programs let you open different types of file and if it's a wavefront it would open the wavefront and if an igram then it would open that etc.

alt-FO  is the secondary "menu" method shortcut where alt+f opens the "file" menu and "o" (with alt still held typically) would open the file.  This works on many versions of linux, and windows back to XP and older and is similar on MAC (command-O I think?)

If you implement "I" it would be nice to also implement CTRL+O as I have hit that key combo many times in DFTF and I can't even tell you if it works or not.  It works in:
notebook, word, excel, chrome, paint...  Pretty much any program that opens and saves files will use CTRL+O to open and CTRL+S to save.

githubdoe, 2025-04-21:
Ready for testing.

gr5, 2025-04-25:
What did you implement?  I tried ctrl+O, I and alt+f.  

Oh - actually if I hit "i" it instantly opens a file called "./.png" which was sitting in the build folder.  I don't think I put that in there? Anyway,  I thought it would pop up the "file open" dialog just like if you do "file" "load interferogram" does.

gr5, 2025-04-26:
Oh - I found it.  It's the "L" key.  What does the "I" key do?  Remove that feature?  Is that your own private debugging feature?  Is it hard to make it CTRL+O?  I suggested ctrl+o because that's just a natural thing for people to try subconciously.  No one will know about the L key or if they learn it they will probably forget it.

githubdoe, 2025-04-26:
Go look at the hot key help.  You will see a note there about looking at the preferences.  I'm not saying more because that doc is supposed to be enough for you to figure it out.

gr5, 2025-04-28:
Okay I found that.  ctrl+o doesn't do anything.  But the hotkey help is nice.

Let me know which of the 5 or so simple bugs I found you don't want to fix.  I could potentially fix most of them myself I suppose.

githubdoe, 2025-04-28:
Sorry I'm a little busy with other things but I will do the fixing.

Here is the reason for I crtl-o and L commands.  I wanted something that would load the latest igram with one finger press.  That was I for Import like when someone imports video in various programs either from a file or from an attached camera.  At the time I knew not all would care for it but would like a one finger press hot key to open the interferogram directory.  So I made it configurable in preferences.  Also needed a config to set where the latest igram might be.  For me that is in the usual windows download directory because it is coming from a web page operated device.

You liked ctrl-o so I added that but I can't press cntrl-o with one finger or even two fingers of my left hand.  Right hand is busy with the mouse.

Eventually I also liked sometime to open the usual igram directory with one finger.  So I added the "L" hot key to do that.  Now that probably is more what ctrl-o should trigger.  The "L" hot key required a new routine and ctrl-o can call it instead triggering the same routine as "I".

I did hook up the ctrl-o and it works on my machine with the code you are testing.  So I'm puzzled why you see nothing happen.   It should do exactly what the I key does.

gr5, 2025-04-28:
Oh.  I want ctrl+o to do what the L key is doing then. It sounds like you use less shortcut keys than I do but I don't even have to think about some of them.  Or rather I just do them subconsciously and don't realize it until one of them doesn't work.

So yes it takes me 2 hands but I don't have to think.  I don't even have to look down at the keyboard.  I don't have to consciously think at all other than - okay let's open another igram.  Same with copy, cut, paste, file save, file save as, select all, page up/down go to start/end of line, search, search-again, print (ctrl+p), close window, zoom in, undo, redo, and much more (truly - much more - for a few years I had RSI (hand injuries) and used only the keyboard for 99.9% of the time and learned that it's better/faster/easier).

These don't all work in all apps but they work in so many that I don't think about it until I've already done one and it doesn't work.

No rush to fix the bugs!  I just wanted to make sure you got all my notes.