githubdoe/DFTFringe #180: Current version crashes when in Ronchi and Foucault view when you change the ronchi ROC offset.

githubdoe, 2025-04-05:
Current version crashes when in Ronchi and Foucault view when you change the ronchi ROC offset.  

Set Foucault ROC offset to .01.  Change Ronchi ROC offset to something bigger than it is.

Crashes in zernike process null_unwrapped() with out of range access to m_rhoTheta.row(i)  where I is iterating over m_row values.  So m_rhoTheta.row and m_row must not be the same size at this point.  Don't know if it should be or not or is the code using a bad indexing value for rhoTheta.  Can't remember the details of this codes intended integrations.  I think rhoTheta should be the same size as the number of points which I think m_row has.  So they have gotten out of sync.  Perhaps in the code that tries not to compute new rhoTheta values to save time.

Will look into it more in the Future unless someone else has already discovered this issue and fixed it and I don't have that fix.

gr5, 2025-04-05:
I didn't look at the code just now but this sounds very much like a bug I fixed in the last year or so.  Are you using the released code or the code from github?  Are you sure you have the latest code?  Did you do a git pull?  Did you merge master into your branch?

I have messed with ronchi offset.  A lot.  Hundreds of times with version 7.2.3.  Never had it crash.  Just played with it again just now.  Seems okay to me.

What if you save your wavefront so you have a consitent starting point, restart DFTF, open the wavefront (only the single wavefront) and don't mess with the resolution, does it still do it?

These crashes I would expect to be related if you change array sizes so if you switch between anulus mode, change resolution, or do a smoothing funciton.

I usually do smoothing, by the way, before messing with ronchi or foucault tab.

githubdoe, 2025-04-05:
I thought I had the latest.  Been too many months gone by since I last tried to update my code which was done last year when I thought I had merged my changes with the master.  But of course how I did that and how to do that usually is a Github mystery to me.    What commands can I use to compare your changes with what I have?    I have made a lot of changes in the last 6 months so a general diff would be too much to dig through.  So if we can limit it to the file that you changed to fix that I can check.  

When I try the published 7.2.3 it does not crash so.....

gr5, 2025-04-05:
I just took a look at what's on github and maybe you are fine:

It looks like you've been working on a branch called "dalework" and the last time that merged into master was 13 months ago on march 13.

However you *did* merge master into your branch most recently last october and that included v7.2.3.  The only things you didn't get were some changes to the readme file and changes to the build process all by Julien.  Dalework has the latest code.

That all assumes you are on the dalework branch.  As long as you are on "dalework" branch you have all the code changes that went into 7.2.3.  If you are on "master" make sure you do a git pull.

githubdoe, 2025-04-06:
Ok, but then if I have the latest it crashes yet reminds me of the crash George fixed.
Anyway,  I think I'm through with all my latest changes.  Mostly about percentage correction and a few minor bug fixes.  Also some new additions like hot key help in the help menu, and auto collimation setting in Ronchi/Foucault.  

George when you get a chance try out that and let me know what you think needs fixing since it may not look good on small resolution displays that I have a hard time testing.  I'm sure there is a bunch of stuff that could be better.  Oh,  I did want to add more info in the help for percentage correction.  I will do that.

githubdoe, 2025-04-12:
Crashes if you press the ROC multiplier two times quickly so that the second press was done while a new ronchi/Foucault image is being computed.

gr5, 2025-04-12:
Duplicated!!  This has a similar feel to the defocus slider crash which I fixed.  I'll look at this.

gr5, 2025-04-12:
I looked at this.  I didn't check super thoroughly but I think it's the same mechanism that caused the defocus crash.  Here's the problem code in image below.

The problem is basically in part 2 in image below.  in part 1 the arrays are cleared.  In part 2 it processes events and gets a second request to update the ronchi/foucault displays - in there somewhere this function is called again, the arrays are cleared again in part 1 and filled in part 3.  Then it returns just past part 2 below.

Then it fills the arrays again so they are doubly sized.  This causes a mismatched array and causes the crash.

The solution for the defocus feature was to add the if statement in part 2 and to set that boolean flag in the defocus code.

why is that code in part 2 even there?  We could probably get rid of it.  I think it may help update some of the progress bars.  Not sure.  I asked you last time and you weren't sure.  If we get rid of part 2 then this may fix other bugs we don't know about.

Or you can just set that same boolean to true in the ronchi code.  If you asked me to fix it, I'd set the flag in the first line of 
void foucaultView::on_makePb_clicked()
which can be seen on "level 4" of the stack trace in the image below.  I'd clear the flag before returning from that function.

Something like:
```
    zernikeProcess &zp = *zernikeProcess::get_Instance();
    zp.m_bDontProcessEvents=true;

```

And then set it to false before returning.  But ideally you can just get rid of "part 2" in the code below and fix other potential bugs we haven't found yet.

[image]

githubdoe, 2025-04-13:
I tried commenting out the 2 section of code.  Then pressing the  up button multiplier multiple times did not crash.  But to my surprise after the ronchi simulation was complete I then pressed the down button and it crashed.

I thought that code helped one cancel long operations that ended up in this section.  To that end I tried zernike smoothing because it can take a long time and I guessed it might use this code.  Another surprise was that the code redraws the progress bar but does not cancel.  Sigh.

githubdoe, 2025-04-13:
There is code in Zernikeprocess ::ZernFitWavefront  that checks the cancel of the progress dialog.  However it is commented out.  It tried to throw a 42.  I did not add that code (I don't think)  However I have decided that it should return a vector or zero length.  Then the caller can test for that and know it should abort.  So I have made those changes and am testing it out.

githubdoe, 2025-04-21:
I think this issue has been fixed and is ready for testing.

gr5, 2025-05-17:
Released in version 7.3.1

closed 2025-05-17 by gr5