githubdoe/DFTFringe #181: Test stand removal report has unreadable contour plots on page 3.

githubdoe, 2025-04-07:
May only happen on large screens like my laptop.

Page 3 of that report is supposed to show what the test stand is doing at each rotation.  To do that it shows contour plots of the test stand only surface.  But they are very tiny and only the scale shows.

[image]

githubdoe, 2025-04-21:
I enlarged the contour plots and the display in the report now.  I also created a log window to show the user the progress.  Also added several processevents so the graphs will update with the progress.

I think this is ready for testing.

gr5, 2025-04-25:
Bad news.  It doesn't look so good for me.

So it pops up a window and the 3rd tab also has it looking just like the pdf.  Before I open the pdf I can make the window bigger and the report looks great but the pdf has already been created and here is a screenshot of part 3:

[image]

gr5, 2025-04-25:
Now if the window that opened was wider then I think the pdf would have been okay.  it opens 800 pixels wide (I have DFTF at full screen size which is 1920X1080).

I have to open that window to 1040 pixels wide for the images to not overlap like the above image.  I don't know if that would fix the pdf or not but since the pdf seems to match the window on the screen so strongly I figured that might help?

githubdoe, 2025-05-06:
Eventually I solved the same problems in the PDF production for the report I think.  I may need to do that again here.  
Yes I could never find a way to get the correct or set the desired width in pixels of the pdf page.  IRRC.  The report may have a key on how.  I think what I did was let the user set the size of the diagrams.

I will fix the other bugs and may leave the PDF problem as a todo.  I have also just discovered another HOT KEY that I forgot to document but have now in new code.  The space bar will toggle the edge zoom feature after the feature has been enabled and in some other cases.

githubdoe, 2025-05-06:
Here I'm going to document what I think may work but don't have time to try. 

You can only get the width in points 1/72 of an inch of the PDF page but at least you can get it.
You can also set the resolution to pixels per inch.  Usually 1200.
So after setting the resolution then you do know the width of the page in pixels.
I think you can use this info to then scale any current window or image to fit how you want on the page.

If I already knew that and tried it but it did not work I can not remember.  So I will try that again.  If it does work then all the other report dialog could be simplified.  But why waist the effort?  Yet we do want to fix the stand removal report.  That last section is one of the most important visual clues of the process's success.

gr5, 2025-05-07:
I'll have time today.  I'll try it.

gr5, 2025-05-08:
Okay I fixed it.  Can you please change the code in your branch?  It's embarassing how long it took me to figure out the change.  Please run the tool to make sure it looks okay on your computer but I'm pretty sure it will be fine.  Below is a screenshot of part of the pdf.  Basically I reduced the size of the last sets of images.  The zern one, the waverfront one, and then the ones for each rotation.

Here's my change. In surfacemanager.cpp, line 1924, change the two 9's to 6's:
old code
```
const int Width = 800 * .9;
const int Height = 600 * .9;

```
new code
```
const int Width = 800 * .6;
const int Height = 600 * .6;
```

[image]

gr5, 2025-05-08:
oh wait!  I also made this change, it's not a huge deal as anyone can resize the window but if I make it 1200 pixels wide by default instead of 800 then the window that pops up (and is almost identical to the pdf) comes up at a reasonable width.  I'm not sure that everyone will realize they can resize the window (probably they will know how to do that).  So it's not critical but I thought an improvement.  Change the line with the "-" to the line with the "+".

```
     tabw->addTab(page3res.Edit, "Page 3 stand analysis");
-    tabw->resize(800,600);
+    tabw->resize(1200,800);
```

githubdoe, 2025-05-08:
Sorry George but your .6 fix is a simplistic fix that does not work for my screen.  Now those images are too small.

gr5, 2025-05-08:
Rats.  What about in the pdf?  They look great in the pdf, right?  I'm attaching the before and after of what the pdf looks like

[stand800x.6.pdf](https://github.com/user-attachments/files/20107461/stand800x.6.pdf)
[stand800x.9.pdf](https://github.com/user-attachments/files/20107462/stand800x.9.pdf)

Are the images at the bottom even smaller than the images at the top.  Let me clarify.  Let's say you used 4 rotations.  Then you should have 8 (quite small) images at the top for the 4 rotations, then one large image. then the x,y astig graph and then the final 6 images are the ones that are too big to fit in the pdf as hopefully you can see in the above ".9" file but fit nicely in the ".6" file above.

githubdoe, 2025-05-08:
Sorry George.  Maybe you should let me fix it.  It is taking me as much time to respond to your questions that I don't have at the moment.

githubdoe, 2025-05-13:
I'm not going to get any time to fix this for awhile.  Maybe we just leave it and publish a version.  I have just updated how one can move a profile up or down when multiple profiles are shown that I like a lot.  I find that a useful way to compare different wave front profiles by making them all zero at the same spot.    Without that they all have varying y offsets to place the mean near the zero line.

gr5, 2025-05-13:
That's fine.  I want to be helpful (by testing) and not slow down releases (much) and I don't want to make you more stressed or anything.

You added another new feature (astig polar graph?).  And also this thing about moving profiles?. If you are ready to release, please update RevisionHistory.html.

Is everything else done?  I'll test out the astig polar graph feature.

githubdoe, 2025-05-13:
Yes I also updated revision history already.  However I was surprised to find that the last few releases are not in it.  I could not find anyplace that had more recent one.

gr5, 2025-05-13:
I love the new features.  Being able to drag a waveform up and down is a delight.  The user interface is so simple.

I like the astitg plot also.

Here's the plot from Lalith's recent wavefronts.  I see you sorted by the angle.  I'd prefer you used the order that the wavefronts are already in.  But the feature is nice as it is.

[image]

githubdoe, 2025-05-13:
I don't sort by the angle.  It is listed by the order they were selected in the list when asked to give a list of the selected items.

githubdoe, 2025-05-13:
The profile drag was an easy implementation copied from the QwtPlot examples on selecting things.  The example is called events.  Which has a class called picker.  I used it to create the profilePlotPicker class.   I just now wished I could also use the scroll wheel to change the y scale of the profile plot.  I bet there is an event for that as well.  However feeding that into the contour plot tools to change the y scale is a bit harder.  But slots and signals could implement it.  Perhaps profile plot connects to the picker scroll signal then sends some info to the m_tools wave front scale functions.

githubdoe, 2025-05-14:
Just implemented scroll wheel scaling.  It was so easy.  I will try to merge soon.  When I discovered the revison history out of date I tried merging but it did not update.  Then I thought I checked various branches to see if there was a newer version but found none.  IIRC.  Maybe I don't know what is the main branch.

githubdoe, 2025-05-14:
OK.  Well I had the latest but thought there was version higher.  The revision history I checked in is an updated version of the latest.  However I called the updates version 7.3.3.  So when the latest version is ready for release that heading probably is not correct.  
I also think I have the latest and pushed that into my branch but of course I'm a GIT dummy so lets figure out if that is correct just as soon as I check in my last tidbits.

githubdoe, 2025-05-14:
All my latest work has been committed and pushed.

gr5, 2025-05-14:
Oh.  So you had the right version after all.  No mistakes.

But can you merge in master?  You have to do something like the following (how I would do it from command line):
```

git checkout master   (switches to your master tracking branch - hopefully you have a tracking branch already.  Most likely.)

git pull             (updates your local master from the github one)

git checkout dalework      (go back to your branch that you were in just before all this)
git merge master               (merges master into your branch)
push dalework
```

If you want me to do it I can but you'll have to do a "git pull" from your dalework branch right after I check it in.

gr5, 2025-05-14:
Also you didn't mention the astig polar plot in the release notes.  You want to add that?

gr5, 2025-05-14:
Also I'd prefer you make it 7.3.0 if it's not too much trouble.  I mean I suppose 7.3.4 is fine but don't you think 7.3.0 is better?  Of course I might end up making a few quick releases if I can't get the installer to work right or something but I prefer 7.3.0 at this point.

githubdoe, 2025-05-14:
Thanks,  I agree and will do about the changes.  Also will try the merge.  Will wait to see if you find anything else.

gr5, 2025-05-14:
> I don't sort by the angle. It is listed by the order they were selected in the list when asked to give a list of the selected items.

Oh.  I played with it some more.  What I did is selected the last one (well it was already selected) and then shift clicked the first one.  So the order was 4,1,2,3.  But if instead I click the first and shift-click the last then it works fine.  Or if I ctrl click each one in order (who does that? lol).

gr5, 2025-05-14:
Okay so I played with your zoom scroll.  It's fine.  

1. Just do the merge from master into dalework and then let me know (here or through an email).  Also please update versionHistory with your polar plot if you haven't yet.
2. I will then create the pull request and send you a link to approve it.  
3. Then after you approve it I will create a new revision tag to match whatever is in VersionHistory and that ill start the release process and the tag will get inserted into the version everywhere.  Then I will test the hell out of the release package (test installer, test the version it installs) and if it all seems fine I'll finalize the release.

githubdoe, 2025-05-14:
Ok master and dalework merged and dalework pushed

githubdoe, 2025-05-17:
Trying to get the astig report better.  On my machine in 4k resolution the windows version of the report has all the images too small to be useful.  Turns out the pdf is the same way.  That is because the report is created based on pixel size of the screen.  It thinks there are n pixels per inch.   I have so many more pixels per inch that a 800 pixel wide image is much smaller than on a 1080 screen.  So I need to rework how the size of each image is created to make them bigger based on screen resolution or independent of screen resolution.

githubdoe, 2025-05-20:
Here are my ideas about this report and how to make it fit any screen size.  I want to make the report window and pdf independent of size of either.

The .pdf is made from the multi tab widget displaying the report.  So that widget is the source and must also be formatted correctly.  Most screens are wide enough to display the report in the same size as the printed page.  So make the images on the screen the same size they will be in the report.  Not trying to support phones here.

However every user can have a different pdf printer page size so that widget should be based on the page width of the current printer.  

Get the page width and make the images that will be placed on it in the same proportion.  Usually slightly under 1/2 with page width for images placed in two columns.

Make the images the same size in inches as the image width calculated from the page width.  You can get the DPI of the current screen so that is easy to compute as desired size * DPI.

Then when the time comes add the images to the html that creates the pages.  

I have implement the last section of the last page of the report using this technique and tested it on my machine and I get a very nice report and pdf file.  Will work on all other pages and images to do the same.  Will test on lower resolution displays later but I think it will work as well.

gr5, 2025-05-20:
In html, you can size images to a different pixel width than the file size and the browser adjust accordingly.  You can also set any component width (including an image) to a percentage of the width of the screen.  Maybe you can use that feature (setting image width percentage) in html and maybe then you don't need to worry about page width.  Just a thought.  Probably won't work anyway.

So for your pdf, I think you picked a default printer, not the printer that the user is using.  And you picked the paper size as A4 I think.  The paper size of A4 set all kinds of things page width and height.

Anyway, this might not be helpful.  Good luck.

githubdoe, 2025-05-21:
Trying to adjust using html scaling was a disaster I once tried.  The problem is that the images made using my computer resolution are way to big if using the lower resolution.  If then instead using the lower screen resolution like you have creates tiny images on my machine.  Thus the change to using image width in inches and adjusting the image size in pixels to make that happen on any machine resolution is the key.

githubdoe, 2025-05-22:
All work is complete on stand removal report and pdf.  Added astig polar plot to it as well.  Test on two resolutions of 4K and 1920x1080.   Made some changes to astig polar class so that the stand removal can call it and create a plot.

Some contours plot as ovals.  That is a hard thing to fix because the contour plot itself is not a square but a rectangle and the size that creates a round disk is not immediately obvious.  Sometimes reducing the height by a factor of .8  from a square creates good round images but not always.  I'm not going to waist any more time to sort that out.

gr5, 2025-05-22:
Dale you haven't merged master into dalework.  There were a few edits by Julien that you didn't get.  Mostly some compiler warnings were fixed.  Also a non-critical memory leak and a critical update to the compiler version for the github build process to work - something to do with armadillo  (doesn't affect DFTFringe_dale.pro or whatever it's called).  Please do:

git status (make sure you've committed everything in Dalework)
git checkout master  (switch to that branch)
git pull
git checkout dalework  (switch back to your branch)
git merge master          (merge master into dalework)

gr5, 2025-05-22:
Two minor bugs
1) When I double click any row twice it doubles up the items in the key as you can see circled. (first time it works perfectly but when I double click it again to see all the astigs the key doubles in size)

2) Also if I click a column header in the table it changes the sort and now if I double click anything in the table it affects the wavefront that was in that row previously which makes things very confusing. So for example if the blue one was first and I resort and now the yellow is first and I double click the first row (yellow) it highlights the blue one (which used to be first row).

The new double-click-to-highlight feature is nice.  And it selects it in the result tab as well.  Nice.  I also like that this graph is in the latest astig-test-stand-removal report.

[image]

githubdoe, 2025-05-22:
I did pull master just a couple of days ago.  Just after you posted that email about how to do it.

githubdoe, 2025-05-22:
I just tried 
git checkout master and it said I was up to date.  Which is what it said a few days ago when I tried.

githubdoe, 2025-05-22:
double clicking adding keys is disappointing.  All I do is change visibility of the unselected series from false back to visible.  At present don't know how to keep that from happening.

I really wanted to select on the graph and have it highlight the wave front.  But that is more complex and I was out of time to implement.  Probably best is to just click on the little disks at the end of the line.  The operation requires a test to see which x,y of the mouse is closest to what thing and if in range of x pixels trigger a select.

Sorting rows changing later changing the selected is also disappointing.  Will look at the cause.

gr5, 2025-05-22:
random suggestions that might give you an idea:

It would be very cool if hovering over the table highlighted a plot instead of double click.  Or single click would be better than double click as well.

What if you just made the highlighted plot line thicker?  Or gray?

githubdoe, 2025-05-23:
Changed selection method to hover over the plotted line.  Specifically the right end point. It will select something if over a line or the left end point but not as specific as the right end point.  Added help text at the top of the plot to point that out.

gr5, 2025-05-23:
It's so much better today.  Sorting doesn't affect the row clicking.  You now highlight with dashed lines.

The hover text is better than the way it was but because it seems to use tooltip it is weird.  sometimes it works fine but then it gets in a weird mode and I have to click somewhere else to get it to work again.  Also tooltips have this annoying delay.  I'm kind of aware of the tooltip weirdness and I assume most people are but I had no idea this was tooltip related until I looked at the code.  Now that I know, it's easier to deal with the weird timing issues.

So I think maybe just a single click would be better than hover?  I don't know.

But I admit that everything works better than before now in this dialog.

I'm going to do more testing later today.

githubdoe, 2025-05-23:
The code tricked you into thinking it uses tooltip.  It does not.  It is just the name of the routing nothing more.  Named that way because the example I used created something they liked to call a tool tip but was not implemented by a Qt tool tip.  They were showing how you would implement something like a tool tip.

gr5, 2025-05-23:
Okay.  Well this hover feature seems to work intermittently.  It gets in a state where it stops working but if I click away and come back it recovers.  For the most part it works fine. I guess I consider this feature done.  Unless you want to try single click?

Once this and the null-stored-in-wavefront-file feature is checked in do you want me to make a new release?

gr5, 2025-06-15:
Fixed in version 7.3.3

closed 2025-06-15 by gr5