githubdoe/DFTFringe #182: Change the way Foucault ROC offset is computed when new wave front is in place.

githubdoe, 2025-04-12:
The ROC offset was set to null the 70%  zone whenever a new wave front was set to be displayed by the Foucault feature.  I do not like this anymore.  Instead I prefer it to be set to what ever was used for the last Foucault simulation.  That makes it much easier to compare different wave front at the same ROC offset.  That is something I have been doing a lot of lately.  

So I made that change and it will be in Dalework branch.

githubdoe, 2025-04-21:
I think this is complete and ready for testing by others.

gr5, 2025-04-21:
I'll look it all over today or tomorrow. You can create a pull request - it will probably automatically default to the right branch.  Or after I look over everything and test it I'll create the pull request.

I did play with it a week or so ago and it all seemed good.  Even the feature where it lets you read wavefronts from multiple directories.

gr5, 2025-04-22:
Dale I can build both release and debug mode but only the debug version launches.  I get an error when I try to run the release version.  Have you tried to build and run in release mode?  Maybe something is wrong with the project file?  The changes you made were pretty minor - I just see where you added new code.  Nothing to do with libraries.

githubdoe, 2025-04-22:
Yes I always do.  Works fine for me.  I did play around with my project awhile back trying to disable logging stuff in Main.cpp.  I don't know if that is causing troubles but it would not be different for debug than release.  Anyway just something to check out.  I'm not sure I ever got it to work the way I wanted with ifdefing a dale specific label to disable the more elaborate logging you all added.

githubdoe, 2025-04-22:
I get crashes on doing various other things after having loaded several wave fronts.  For instance just changing the vertical resolution on the 3D controls.  But not always I think.   Ouch.  Need to do more analysis but just a heads up.

githubdoe, 2025-04-23:
But after a full day of using DFTFringe it never crashed again after about 2 crashes earlier in the day.  So.....I don't know the sequence needed to make it happen.

gr5, 2025-04-25:
I prefer the way it is now in branch dalework.  I agree that you successfully keep the ronchi/foucault ROC Offset from jumping to the "70% zone" value.  However...

How do you calculate the 70% zone?  Everytime I click the "70% zone" button, DFTF just sets the ROC Offset back to 0.  I guess I never understood that button.

I'm testing using an igram with a null of -0.5.  Diameter of 146mm (roc 1587).  Why does 70% zone button always give me 0 for the ROC Offset?

Oh wait!!  That's if I have it in mm.  If I have it in inches it seems to work (0.034 inches).  70% zone should have set my roc offset to 0.8636mm.

githubdoe, 2025-04-25:
That button is supposed to set the ROC offset to a value that will null the 70% zone.  Easy to compute from the formula of the conic  using F number and diameter and desired conic.   

It probably forgot to check units and adjust accordingly.  Will look at it.

gr5, 2025-05-03:
great.  I'll recheck the things I found before and let you know.  I'll probably get to it today but if not, I should have lots of time for this Monday.

gr5, 2025-05-17:
done.  Implemented in version 7.3.1

closed 2025-05-17 by gr5