githubdoe/DFTFringe #183: Display astig of selected wave fronts in polar form

githubdoe, 2025-05-07:
This was a request many years ago to display the astig values in polar form.  It is nicer to see the overall magnitude of the astig value when trying to compare various figuring steps with each other.  At the time I only added it to the report.  That however only shows the astig for the currently selected single wave front.  It also takes time to display it.  

I think it would be more helpful to display some sort of graph which plots the selected wave fronts on a polar plot.

I'm currently working a mirror with a lot of astig and I want to compare techniques used from session to session to reduce the astig.  In my case the x value was reducing in size but the Y value was increasing so it was hard to decide if the magnitude of the astig was being reduced without doing some math outside of DFTFringe.

So I have started working on a polar display of astig that is triggered from the view menu and pops up in a dialog.

githubdoe, 2025-05-08:
Feature is finished I think.  Pushed code to dalework

gr5, 2025-05-22:
This is done in version 7.3.1.  Closing.

closed 2025-05-22 by gr5