githubdoe/DFTFringe #191: migration to QT6

atsju, 2025-05-23:
Hi Dale and George,

As you probably know, Qt5 reached end of life in 2023, and extended support is ending this month. While the source code will remain available, it's likely that the online documentation and related resources will eventually be removed — if not immediately, then probably within the next 24 months.

[image]

Do you already have a plan regarding this deprecation?

On my side, I experimented with a migration to QT6.8 LTS on a development branch to see what it would take. It looks like I was able to complete the migration successfully.

You can try a working installer here:
👉 [Download the installer](https://github.com/atsju/DFTFringe/actions/runs/15204446750/artifacts/3183138911)

Due to dependency constraints, I had to update the compiler, Qt, Qwt, and OpenCV together to make everything work properly.

I plan to open a pull request once I've cleaned up the code and updated the documentation.

What are your thoughts on this?

Dale – to ease a potential transition on your side, I can provide detailed documentation on how to recreate your build environment with the updated dependencies (including OpenCV, where your current version differs from the one used in our distribution).

Alternatively, I can send you a pre-configured folder with everything set up. That way, you could simply open the DFTFringe.pro in Qt Creator, only needing to link the updated Qt and compiler versions.

Looking forward to your feedback.

atsju, 2025-05-27:
Hi @githubdoe, any feedback on this ? I noticed QT5 documentation is already gone from QT website. I'm relieved I made the porting with the documentation still easily accessible.

gr5, 2025-07-15:
I was thinking it wouldn't be hard to write a bash script or a c++ program to automatically convert the entire stack trace.  However it would be quite slow.  addr2line (or whatever the tool is called) can take sometimes 5 seconds per conversion and a stack trace can typically have 20 lines to be translated.

But it also occurred to me that the program itself could do the conversion as part of the log creation.  But having it sit there for a minute while it's crashing would be very annoying.

atsju, 2025-07-15:
> But it also occurred to me that the program itself could do the conversion as part of the log creation. But having it sit there for a minute while it's crashing would be very annoying.

Plus the risk of the process failing. The logging shall be extra safe, thus simple.

atsju, 2025-07-16:
> Note to myself : this might solve the GCC and C++23 version deadlock that made me use boost library instead https://en.cppreference.com/w/cpp/utility/basic_stacktrace.html
> 
> To be tested. It could be a massive improvement if the stacktrace was directly readable without need for translation.

mingw 13.1.0  as provided for the QT version I am using has not been built with stacktrace support. Keeping `boost` for now

atsju, 2025-08-04:
@gr5 in the end there are only 9 source files modified. What do you think about having #ifdef on this ? 
We would then release with QT6 and Dale can continue development on QT5

I can even update CI to build both QT5 and QT6 versions.

gr5, 2025-08-04:
I'm quite curious if qt6 fixes the UFT-8 issue that is still open.

9 files but how many sections of code would need ifdef around them?

githubdoe, 2025-08-05:
Go ahead and change as needed.  I'm just too busy to try and update to any new versions of qt, openCV........  But don't wear yourselves out trying to accommodate me.  When it becomes important to me I will update to the newer tools.  At present I will just stay with the older source code of DFTFringe.

PR #216 (2025-08-07): QT6
Here we go. Fully compatible with Qt5 and Qt6. CI/CD with installer generation for both.
Next release should be named v8.0.0 (no hurry on that).

Build on:
- QT6.8 (LTS) instead QT5
- QWT 6.3.0 instead 6.1.6
- OpenCV4.12.0 instead 4.6.0

Using:
- C++17 instead C++11
- mingw13.1.0 instead 8.1.0

Close #191
closed 2025-08-07 by gr5