githubdoe/DFTFringe #196: Crash after starting compute

KineticPrecision, 2025-06-09:
**Describe the bug**
After importing an image (we tried two), and setting up for compute, the program quits after hitting the Compute button.

**To Reproduce**
Steps to reproduce the behavior:

1. Import image, look at intensity, choose color (green).
2. On the DFT boundary setting, set circle to around center spot.
3. Hit Compute button.
4. Program quits.

**Expected behavior**
Expected a fringe analysis.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - DFTFringe version: 7.3.3
 - Build: from GitHub release

**Additional context**

Found this in the logs:

[2025-06-09 11:55:01.433] [critical] CV error :ranges[i] && ranges[i][0] < ranges[i][1] in D:\a\DFTFringe\DFTFringe\openCV\modules\imgproc\src\histogram.cpp on line 202
[2025-06-09 11:55:01.433] [critical] Unexpected issue. Stacktrace:

Found the same issue by installing and running version 6.3.1.

[log.txt](https://github.com/user-attachments/files/20657564/log.txt)

gr5, 2025-06-09:
I'll translate the log file and see what I find.

That error that mentions the D: drive has to do with where the file was when it was compiled which should have been on some drive on github somewhere I think?  Not sure but the other lines of the stack trace will probably tell me much more.

atsju, 2025-06-10:
Hi. Yes D: is GitHub actions build folder. To translate log see here https://github.com/githubdoe/DFTFringe?tab=readme-ov-file#how-to-analyse-crashlog
Usually only few first and few last lines are useful.

I'm sorry to not help more. I have no access to computer

gr5, 2025-06-10:
Here is the relevant part of the stack trace. In **bold**, I added comments; mostly file and line number gleaned from the addr2line tool.

> [2025-06-09 11:53:55.396] [info] DFTFringe v7.3.3 started
> [2025-06-09 11:53:56.307] [trace] Log level changed to: Info
> [2025-06-09 11:55:01.433] [critical] CV error :ranges[i] && ranges[i][0] < ranges[i][1] in D:\a\DFTFringe\DFTFringe\openCV\modules\imgproc\src\histogram.cpp on line 202
> [2025-06-09 11:55:01.433] [critical] Unexpected issue. Stacktrace:
>  0# 0x00000000005B46C9 in C:\Users\Spencer Webb\DFTFringe\DFTFringe.exe **ignore: these lines create the**
>  1# 0x0000000000464AC5 in C:\Users\Spencer Webb\DFTFringe\DFTFringe.exe **ignore: debug output displayed**
>  2# 0x0000000000464D2F in C:\Users\Spencer Webb\DFTFringe\DFTFringe.exe **ignore: here**
>  3# 0x0000000001749FE0 in C:\Users\Spencer Webb\DFTFringe\libopencv_core460.dll   **(this is the crash)**
>  4# 0x000000000174A1E2 in C:\Users\Spencer Webb\DFTFringe\libopencv_core460.dll
>  5# 0x000000006561CF0E in C:\Users\Spencer Webb\DFTFringe\libopencv_imgproc460.dll
>  6# 0x000000006552D50C in C:\Users\Spencer Webb\DFTFringe\libopencv_imgproc460.dll
>  7# 0x000000000049BB2C in C:\Users\Spencer Webb\DFTFringe\DFTFringe.exe **pixelstats.cpp:518  [point B]**
>  8# 0x00000000004E0602 in C:\Users\Spencer Webb\DFTFringe\DFTFringe.exe **surfacemanager.cpp:699**  
>  9# 0x00000000004E0D4E in C:\Users\Spencer Webb\DFTFringe\DFTFringe.exe **atomic_base.h:396**
> 10# 0x00000000004E0E2C in C:\Users\Spencer Webb\DFTFringe\DFTFringe.exe **surfacemanager.cpp:1424**
> 11# 0x00000000004E408B in C:\Users\Spencer Webb\DFTFringe\DFTFringe.exe **surfacemanager.cpp:1084 [point A]**
> 12# 0x000000000052F2BA in C:\Users\Spencer Webb\DFTFringe\DFTFringe.exe **moc_surfacemanager.cpp:297**
> 13# 0x0000000068B978BB in C:\Users\Spencer Webb\DFTFringe\Qt5Core.dll
> 14# 0x00000000005230DE in C:\Users\Spencer Webb\DFTFringe\DFTFringe.exe **moc_dftarea.cpp:316**
> 15# 0x000000000043A16E in C:\Users\Spencer Webb\DFTFringe\DFTFringe.exe **dftarea.cpp:1053**
> 

Above line of code (15) is called when you click "compute surface"
Everything in the rest of the stack trace (16-46) I removed because it is basically code not written by dale but created by QT to handle the mouse click

How it gets from point A above to point B above seems strange.  I think that part of the stack trace may be messed up?

Anyway I'm quite curious if you ever clicked the "Histogram" button between when you started the program and when it crashed.  If you didn't then this is going to be hard to reproduce.  If you did... then I think it's also going to be hard to reproduce.

gr5, 2025-06-10:
He means "windows registry".  You can run "regedit" on windows.  Search for DFTFringe and find something that looks like this screenshot below.  Right click the dftfringe key folder, choose export and post that file (it's a text file).

[image]

gr5, 2025-07-11:
I'll translate this into line numbers but I can't look at it probably until tomorrow.  Tonight at the soonest.  I see it's version 7.3.4.  There's a tool to translate the numbers to line numbers using the debug file that is on the releases page for dftfringe.

gr5, 2025-07-12:
I'll try to duplicate that.

gr5, 2025-07-12:
Wow - it's crashing on the exact same line of code: pixelstats.cpp line 518.  Even though this newer crash is version 7.3.4 and the older crash at the top of this topic was 7.3.3

Yeah it's something really weird that is messing with the stack so we can't tell quite exactly where it is crashing.  I only annotated 3 points on the stack trace but they are the same lines of code as above so I stopped after 3.

 0# 0x00000000005B4909 in D:\Programs\dft\DFTFringe.exe **boost/stacktrace (code to display this stack trace)**
 1# 0x0000000000464AC5 in D:\Programs\dft\DFTFringe.exe
 2# 0x0000000000464D2F in D:\Programs\dft\DFTFringe.exe
 3# 0x00000000680F9FE0 in D:\Programs\dft\libopencv_core460.dll **(this is the crash: CV Error)**
 4# 0x00000000680FA1E2 in D:\Programs\dft\libopencv_core460.dll
 5# 0x000000006561CF0E in D:\Programs\dft\libopencv_imgproc460.dll
 6# 0x000000006552D50C in D:\Programs\dft\libopencv_imgproc460.dll
 7# 0x000000000049BC6C in D:\Programs\dft\DFTFringe.exe **pixelstats.cpp:518**
 8# 0x00000000004E0742 in D:\Programs\dft\DFTFringe.exe
 9# 0x00000000004E0E8E in D:\Programs\dft\DFTFringe.exe
10# 0x00000000004E0F6C in D:\Programs\dft\DFTFringe.exe
11# 0x00000000004E4274 in D:\Programs\dft\DFTFringe.exe
12# 0x000000000052F50A in D:\Programs\dft\DFTFringe.exe
13# 0x0000000068B978BB in D:\Programs\dft\Qt5Core.dll
14# 0x00000000005232DE in D:\Programs\dft\DFTFringe.exe **moc_dftarea.cpp:316**
15# 0x000000000043A16E in D:\Programs\dft\DFTFringe.exe
16# 0x0000000000523CF7 in D:\Programs\dft\DFTFringe.exe
17# 0x0000000068B978BB in D:\Programs\dft\Qt5Core.dll
18# 0x0000000000524807 in D:\Programs\dft\DFTFringe.exe
19# 0x0000000000524999 in D:\Programs\dft\DFTFringe.exe
20# 0x0000000068B975ED in D:\Programs\dft\Qt5Core.dll
21# 0x0000000001614BB2 in D:\Programs\dft\Qt5Widgets.dll

gr5, 2025-07-13:
By the way, I was not able to duplicate the problem.  I tried following the directions in the first post with version 7.3.3 and 7.3.4.  I also tried running from a d:\ drive and also I tried from a v:\ drive that was a network drive on a NAS.  I couldn't get it to crash.

If someone can get it to crash repeatedly starting with a fresh launch, please make a screen capture video of the process.  Also please send me the igram you are using.  Also (least important) please dump out your registry values - find the values as I show them in my 3rd post from the top (3rd post by me).  Then right click on the "folder" and do "export" and post the ".reg" file here.  Oh and attach the crash log from the log file similar to the post just above this one (but you don't need to do the translation part I did above).

atsju, 2025-07-14:
@gr5  I'm able to reproduce the first one from jagermeisterlover on 7.3.4 
- I installed the official release
- open DFTFringe
- do nothing except : tool -> wevefront inspector
Application crash with no information in stacktrace.

[log.txt](https://github.com/user-attachments/files/21220288/log.txt)
[reg.reg.txt](https://github.com/user-attachments/files/21220386/reg.reg.txt)

I have attached my registry, but I have been able to reproduce with registry entry deleted (this means like a new computer).

I can probably investigate this as I can both build and reproduce but I won't have too much time before some days.

The absence if stacktrace is worrying me. Discussion with chatGPT told me we should use SetUnhandledExceptionFilter() in additon to   std::set_terminate for windows machine. And the we could MiniDumpWriteDump to have a minidump in addition to stack trace.

gr5, 2025-07-14:
Oh.  Sorry.  That one I can definitely reproduce.  The fix is probably to keep it from launching at all or to check if there is a wavefront at all.

I meant the other crash where we have 2 stack traces above.  I can't reproduce that.

atsju, 2025-07-14:
I will try when I have time.

Keep from launching is what I would call a workaround and is step 2. Ideally step one would be an explicit and safe failure with log.

gr5, 2025-07-14:
It doesn't make sense to view the wavefront if there is no wavefront to view.  And when you first launch the program, there are no wavefronts.  So we could:
1) Make the option gray and disabled if there is no wavefront to view.
2) Have it pop up a message that says you need to have a wavefront before you can use that feature.

There are potentially many other bugs that happen when you try to do something before you've loaded or calculated a wavefront.

githubdoe, 2025-07-15:
You may notice that when you had wave fronts and you deleted them all it creates a dummy wave front.  It use to do that at startup.  I found that to be a nuisance since then you can do things with all of the future wave fronts that are loaded but  it is polluted by the dummy wave front.  So I removed that generation on startup.  It did however keep many wave front processor bugs from happening when there were no wave fronts and I was too lazy to add check to each.   It of course is not hard to add those checks.  It is just tedious and not very productive.

gr5, 2025-07-17:
This is quite interesting and may indeed have found a bug.  But I'm very confused about why it's calling updateHisto at all?

Users are not reporting that they are selecting the histogram feature in the first place.  This is an obscure feature.  And if you look at the stack trace it suddenly jumps into updateHisto() for no apparent reason?

Or maybe I made a big error somewhere and it does get called during normal processing?

githubdoe, 2025-07-17:
IIRC there is only one histogram and that is what makes the PV pixel histogram from above the contour plot.  I don't remember any of the specific parameters and don't have time now to investigate.  If it only happens where there is no wave front then just don't let it happen with no wave front.

githubdoe, 2025-07-18:
The comment in the code about hue and sat are probably left over from the example I copied.  Sbins are how many vertical bars you want in the histogram.  So yes it is important.

gr5, 2025-07-18:
> Sbins are how many vertical bars you want in the histogram. So yes it is important.

That's not how calcHist() works.  It's confusing and I read the documentation a few times.  AI is right, you only need the first set of bins (1000 bins).  A 2D histogram would be displayed in 3D which we don't do.  It makes sense to me now.  Why it sometimes crashes and why "solution A" is the way to go (although I don't think the code will compile the way the AI wrote it - very strange that it figured out the problem exactly but the fix is bad as well and will get a compile error).

I'll fix this probably Saturday.  The "solution A" will work pretty much as written.  I'll test it as well of course.

githubdoe, 2025-07-19:
Yes I guess so.  Since I see a large number of bins in the actual display.  Perhaps it is that 1000 number.

githubdoe, 2025-07-19:
I have not looked at it as deep as you have.  I found that maybe all that is needed is to change histat to be one value = {1000}.  Of course delete the sbin variable and the left over comments.  Doing that worked for me but of course it never crashed for me either.

atsju, 2025-07-19:
@gr5 
I just want to share a discovery that will help later debug.

Instead of running `addr2line -e DFTFringe.exe.debug 0x00000000005A7229` for exah line I just run `addr2line -e DFTFringe.exe.debug` in cmd. Then I paste all the adresses at once (use a tool capable of multiline edition to copy only address). No need for a script.   
Also note that on my side, addr2line is very fast.

atsju, 2025-08-03:
That should be fixed in v7.4.0. I'm closing it now but do not hesitate to open again or a new issue in case of crash

closed 2025-08-03 by atsju