githubdoe/DFTFringe #199: It would be nice to reset the auto invert wave front condition

githubdoe, 2025-06-17:
I have been working on a mirror that is mostly a sphere and so near to one the conic sometimes switches sign.

DFTFringe will ask you when that happens if you have a conic set other than 0.  Once you say yes it will no longer ask you until you restart the program.  That usually is a convenience.  However in my case there are times I don't want it to invert the wave front but did want it to earlier.  

It would be good to have a way to reset that auto invert flag.  I have added a way to do that in the tools menu in dalework.

Sorry for more work.

Dale

gr5, 2025-06-18:
Please see issue #167 

Can we do that instead?  I'm happy to write that code if you like the idea.  Please provide feedback over on that issue.

Also it seems to me the "invert" button should reset the "auto invert flag".  That should satisfy your need above.

gr5, 2025-06-19:
PR 201 is merged with master.  I'm going to do this feature - at least enough to demo it in a video and get annoying feedback some of which will hopefully be helpful.

Probably won't work on this for 2 days as tomorrow I'll be busy.

gr5, 2025-11-11:
Well you can get to it from the bottom left corner of the mirror dialog now.

but I'd of like to keep the menu item that resets things.  It's helpful for testing as well.  "tools" "stop auto invert".  It's fine as is.  This ticket can be closed.