githubdoe/DFTFringe #224: refactor QSettings usage

atsju, 2025-08-14:
## Proposal for 8.0.0 release
I propose to create a `settingsManager` class that would abstract `QSettings`. 
Every place where we use `QSettings` we would instead use `settingsManager::getSomethingValue()` and other dedicated method to read/write each of the possible settings.

I would use this occasion to reorganize keys into a tree and rename keys for more clarity. Especially use same keys for json and QSettings.
I would also add documentation.

## Advantage

- Centralized listing of all QSettings.
- More maintainable in long term
- No risk of using wrong key or different default values in different places.
- Tress structure for easier parameter tracking in regedit

## Disadvantage

- Users will lose most of their settings moving to 8.0.0
- Each read/write will do a fresh instance of QSettings. Impact should not be too heavy.

## Code example: 
```C++
#ifndef SETTINGS_MANAGER_H
#define SETTINGS_MANAGER_H

#include <QString>
#include <QSettings>
#include <QVariant>

class SettingsManager {
public:
    // Delete constructor to prevent instantiation
    SettingsManager() = delete;

    // Example: Get a value of "something"
    static int getSomethingValue() {
        QSettings settings;
        return settings.value(somethingKey, defaultSomethingValue);
    }

    // Example: Set a value
    static void setSomethingValue(int value) {
        QSettings settings;
        settings.setValue(somethingKey, value);
    }

    // Add more get_/set_ methods as needed...
};

#endif // SETTINGS_MANAGER_H
```

What do you think ?

githubdoe, 2025-08-14:
If I as a user knew I would loose my settings I would never upgrade until a new feature that I thought was a must have emirged.

I have tone of .ini files in all my old mirror tests.  I would not like to have to go back to an old version (which I may have lost) just to look at those mirrors.  Yes I do go back quite often.

githubdoe, 2025-08-14:
If it ain't broke don't fix it.  Is my first rule.  Especially if it changes something the user will notice in a bad way.  If it is not fixing a bug or adding a user seen enhancement then it should not be considered as a high priority.  Perhaps not considered at all.  Yes I often miss spell a settings keyword when created a save and fetch pair.  Something I have learned to test more when creating and something that causes me more waisted time then I like.  I have learned to make sure I do a cut and paste of the keyword.

I don't like telling users about how to find keywords in the registry by the way.  It confuses some.  Yes there have been times that is the best work around.   But try to limit doing that.  It is not a user friendly place.  The registry.

githubdoe, 2025-08-14:
I do think the preference user interface could stand improvement.  It is rather clunky.  Sometime apply is needed sometimes not.  More pages are needed but it is hard to add them especially since they are widgets.  I have wanted to add more preferences but have not since it is so hard.  I welcome any rework to the preferences GUI.